fpgadeveloper/fpga-drive-aximm-pcie #1: Can't get Linux to boot KCU105

cafeclimber, 2017-06-28:
I'm trying to get Linux up and running on a KCU105 with no luck. I can get all the way through the PCI device enumeration example no problem using the steps outlined in this repo's README. I then switch over to my Linux virtual machine and the instructions for the KC705 from your [blog](http://www.fpgadeveloper.com/2016/04/connecting-an-ssd-to-an-fpga-running-petalinux.html). I make it all the way through configuring and building Linux and attempting to boot over JTAG. I've got a serial console open on my Windows host watching the relevant COM ports at the baud rate and settings specified by your instructions. The commands I use are shown below:

```
$ petalinux-boot --jtag --fpga --bitstream ../kcu105_hpc_pcie_wrapper.bit
$ petalinux-boot --jtag --kernel
```
The first command runs and returns fine. The second command runs and returns, however after waiting for a number of minutes, there is no output to my serial consoles.

After a number of attempts, I tried to boot with QEMU to see if it was the image:

```
$ petalinux-boot --qemu --kernel
```

Which boots fine, however I did notice this line:

...
PCI: CLS 0 bytes, default 32
Skipping unavailable RESET gpio -2 (reset)
futex hash table entries: 16 (order: -4, 448 bytes)
audit: initializing netlink subsys (disabled)
audit: type=2000 audit(2.020:1): initialized
workingset: timestamp_bits=30 max_order=18 bucket_order=0
romfs: ROMFS MTD (C) 2007 Red Hat, Inc.
bounce: pool size: 64 pages
io scheduler noop registered
io scheduler deadline registered
io scheduler cfq registered (default)
Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
console [ttyS0] disabled
44a00000.serial: ttyS0 at MMIO 0x44a01000 (irq = 3, base_baud = 6250000) is a 16550A
console [ttyS0] enabled
brd: module loaded
libphy: Fixed MDIO Bus: probed
NET: Registered protocol family 17
Key type encrypted registered
**Warning: unable to open an initial console.**
Freeing unused kernel memory: 7264K (c0510000 - c0c28000)
This architecture does not have kernel memory protection.
random: fast init done

PetaLinux 2017.1 plnx_microblaze /dev/ttyS0

plnx_microblaze login: 

I'm wondering if this could be contributing to my issues. Any help you'd be able to provide would be deeply appreciated. I can provide the full output from the QEMU boot if necessary

fpgadeveloper, 2017-06-29:
Hi Ryan,
I'll look into this today or tomorrow at the latest and get back to you.
Jeff

cafeclimber, 2017-06-29:
Thanks, Jeff. I appreciate your taking the time!

fpgadeveloper, 2017-06-29:
Hi Ryan,
I just tested it out on my KCU105 board using Vivado and PetaLinux 2017.1, and everything worked fine. Please try running the attached script to build your PetaLinux project and let me know if it worked for you.

(1) Under Linux, you will have to extract the contents of the zip file to your fpga-drive-aximm-pcie repo directory. The Vivado project must already be built and exported.
(2) Then in Linux, cd to the "PetaLinux" directory (from the zip file) and run the script called "kcu105_hpc_build". That will take a while and it will build and configure the PetaLinux project.
(3) Power up the KCU105 board with FPGA Drive FMC + SSD attached. JTAG must be attached.
(4) Run "petalinux-boot --jtag --fpga --bitstream ../../Vivado/kcu105_hpc_pcie/kcu105_hpc_pcie.runs/impl_1/kcu105_hpc_pcie_wrapper.bit"
(5) Run "petalinux-boot --jtag --kernel"

Let me know how that goes.

Jeff
[PetaLinux.tar.gz](https://github.com/fpgadeveloper/fpga-drive-aximm-pcie/files/1112531/PetaLinux.tar.gz)

cafeclimber, 2017-06-29:
That worked! I have no idea what I might have been doing wrong. I took a look at your script and it looks to be what I had been doing (except the oldconfig switch and the copying of the source directory.)

Thanks again for taking the time!

closed 2017-06-29 by cafeclimber