fpgadeveloper/fpga-drive-aximm-pcie #4: Running with KCU105

vglinden, 2019-04-24:
Hi Jeff,

I am trying to run unsuccessfully the **kcu105-hpc-dual** project. I have previously ran this project with the kc705 board before and it worked fine. I have also ran the loop-back test (with both boards) and it also works fine. The problem after I execute all of the steps, there is no feedback on the terminal.

Because the kcu105-hpc_-dual_ did not work, I tried the kcu105-hpc, but when I build it through the _.bat_ file, the _tcl_ console got stuck at `create_bd_cell -type ip -vlnv xilinx.com:ip:ddr4 ddr4_0`. I remember that it got stuck at some line (maybe the same?) when I've tried to build the "-dual" project. Then I force quit and tried again and there I went, just as I have for this non dual project. The single SSD project also didn't work.

I am running on a Windows 10 machine with Vivado 2018.2, but I have access to Linux, with the same version of Vivado installed.

Thank you,

fpgadeveloper, 2019-04-30:
Hi,

I just cloned the repo and build the projects you mentioned without any issues.

Can you tell me the path of your projects? Where are you cloning the repo to on your hard drive?

Regards,
Jeff

vglinden, 2019-04-30:
Hi,

The path is "D:\Linden\fpga-drive-aximm-pcie", just as it was for the KC705 board (which worked). I cloned it through the teminal command `git clone https://github.com/fpgadeveloper/fpga-drive-aximm-pcie.git`.
It does build without issues. But as I mentioned, I get no feedback from the UART.
I am probably doing something wrong here, but I don't know what...

Regards

fpgadeveloper, 2019-05-01:
Hi,

That sounds fine, so maybe the problem is your hardware. What are the serial numbers of your Ethernet FMCs?

Regards,
Jeff

fpgadeveloper, 2019-05-01:
Hi,

I have been able to rebuild the project from a fresh clone and then run the standalone application on my hardware without any problem. Did you set your console to 9600 bps? Please give me more info on your setup.

Regards,
Jeff

vglinden, 2019-05-01:
Hi Jeff,

Aujourd'hui est un jour fériée dans mon pays, puis je n'ai pas d'access a ce moment-là.
Je vous repondrai demain. Ok?

Merci

vglinden, 2019-05-02:
> That sounds fine, so maybe the problem is your hardware. What are the serial numbers of your FPGA Drive FMCs?

The FPGA Drive:
- the serial number is 470309;
- the part number is OP047.

> Did you set your console to 9600 bps?

The console (PuTTY) is correctly set, because I get feedback from the BIST application that comes with the FPGA.

Sorry for the delay, I will keep trying here to fix the problem.

fpgadeveloper, 2019-05-02:
Does the BIST application drive the UART at 9600 bps? Can you try it out and make sure?

vglinden, 2019-05-02:
Hi Jeff,

**You are right**. Despite the normal baud rate for the UART in the KCU105 board being 115200, for this project it is 9600. I would advise you to make a comment in the project wiki about this :D. Thank you!

Sorry for the trouble. Case closed!

closed 2019-05-02 by vglinden