fpgadeveloper/fpga-drive-aximm-pcie #5: ZCU106 ELF downloading failed

yu-zou, 2020-04-24:
When running command "petalinux-boot --jtag --fpga --bitstream ./images/linux/system.bit"
It shows "Failed to download PetaLinux/zcu106_hpc0/images/linux/bl31.elf"

Thank you.

fpgadeveloper, 2020-05-04:
I just tried this on my hardware without any problem. Here is my output:

```
jeffrey@linux-pc:/media/opsero/vivado-2019-2/fpga-drive-aximm-pcie/PetaLinux/zcu106_hpc0$ petalinux-boot --jtag --fpga --bitstream ./images/linux/system.bit
INFO: sourcing build tools
INFO: Launching XSDB for file download and boot.
INFO: This may take a few minutes, depending on the size of your image.
rlwrap: warning: your $TERM is 'xterm-256color' but rlwrap couldn't find it in the terminfo database. Expect some problems.: Inappropriate ioctl for device
INFO: Configuring the FPGA...                                                   
INFO: Downloading bitstream: ./images/linux/system.bit to the target.
INFO: Downloading ELF file: /media/opsero/vivado-2019-2/fpga-drive-aximm-pcie/PetaLinux/zcu106_hpc0/images/linux/bl31.elf to the target.
```
Does your `bl31.elf` file exist? What version of the tools are you using? Do you have the FPGA Drive FMC installed on your HPC0 connector? What is the serial number? Does the bitstream succeed in downloading? Please provide more information.

Jeff

yu-zou, 2020-05-05:
Hi thanks for the reply.

1. bl31.elf indeed exists.
2. I'm using Vivado 2019.2.
3. FPGA drive is installed, and I tested by inserting SD card.
4. ZCU106 I'm using is a newer version, which has different DDR settings from the previous version. Do I need to modify the PS DDR setting before generating bitstream?
5. Bitstream is successfully downloaded.

fpgadeveloper, 2020-05-06:
The newer version ZCU106 should not cause problems because you are using 2019.2. What is the serial number of your FPGA Drive FMC? What is the model and type of SSD that you are using?

yu-zou, 2020-05-06:
The serial number of FMC is: 470354
SSD: Samsung 970 EVO.

I also tried the FMC and SSD on KC705 using Vivado 2018.2. There was no problem downloading using either JTAG or SD.

I will let you know if I find more issues.

fpgadeveloper, 2020-05-06:
Have you tried booting the ZCU106 PetaLinux BSP via JTAG? Does it work or do you get the same error? Have you tried re-building the entire PetaLinux project (delete it and re-run build-petalinux script)? I am not able to reproduce the error, so I suspect this is an issue with your hardware or with your setup.

yu-zou, 2020-05-07:
Could I ask how to boot BSP via JTAG?

I tried deleting and rebuilding, but it did not work.
We have two ZCU106, I will try this on the other one.

Thx.

fpgadeveloper, 2020-05-07:
Download the ZCU106 BSP from here:
https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html

Then follow the PetaLinux user guide for instructions on installing the BSP. That will create a PetaLinux project for the ZCU106 with basic components. You can then try to boot the BSP project with the same command as you are using above.

fpgadeveloper, 2020-05-07:
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2019_2/ug1144-petalinux-tools-reference-guide.pdf
PetaLinux user guide

yu-zou, 2020-05-14:
Thanks for the reply. But I tried that approach, also failed. The sa

closed 2020-05-14 by yu-zou