fpgadeveloper/fpga-drive-aximm-pcie #6: S_AXI_LITE's assigned address is out of range

yu-zou, 2020-04-29:
[image]
As shown the CTRL0 address was assigned to 0x04_0000_0000.
However, the S_AXI_LITE address channel is 32-bit, which is incompatible with assigned address.
[image]

fpgadeveloper, 2020-05-04:
A 32-bit AXI-lite interface can be assigned a 40-bit address as we have done in this design. If you find any documentation to the contrary, please post a link here. If you have a warning message from the Vivado tool, please post it here.

closed 2020-05-04 by fpgadeveloper

yu-zou, 2020-05-05:
That's a stupid question. Never mind.