fpgadeveloper/fpga-drive-aximm-pcie #7: Multiple issues while building the SW project

jdlr-at-nrec, 2020-04-30:
Executing the build-sdk.bat as directed has several issues with dec2019 release for zcu106 dev kit.

1- Warning: No app will be generated for Zynq Ultrascale+ designs. ZCU106 dev kit has a Zynq Uscale+ device
2- ELF does not exist for zcu106_ssd_test. Chking build status: no files matched glob pattern "./_ssd_test"

I am not verse in .tcl and have not solve this yet. :(

fpgadeveloper, 2020-05-01:
Please pull the latest commit. The SDK is not part of the 2019.2 tools so the SDK script has been reworked for Vitis in the last commits.

fpgadeveloper, 2020-05-01:
Also, a software application for the Zynq US+ designs has been added to the latest commit.

jdlr-at-nrec, 2020-05-01:
I don't have access to 2019.2. Will it work with 2019.1?

fpgadeveloper, 2020-05-01:
In that case you'll have to build the SDK workspace manually (without the script). Create the hardware platform for the ZCU106 using the exported Vivado project, then add a new empty application. Then copy this file into the application sources and it should work:
"C:\Xilinx\SDK\2019.1\data\embeddedsw\XilinxProcessorIPLib\drivers\xdmapcie_v1_0\examples\xdmapcie_rc_enumerate_example.c"

closed 2020-05-01 by fpgadeveloper

jdlr-at-nrec, 2020-05-01:
Excellent!

jdlr-at-nrec, 2020-05-01:
Followed your recommendation and worked perfectly under 2019.1.

Thank you for your guidance.

jdlr-at-nrec, 2020-06-09:
Hello Jeff,

Hope you don’t mind but i have become desperate.

I had “fpga-drive-aximm-pcie-2018.2_1” running simulation and i don’t even remember how; but i don’t seem to be able to find the test bench i used. You helped get the design to compile and run on our FPGA Drive FMC and that still runs fine and very happy about that.

Unfortunately during a purge to make it easier to find my stuff, i deleted the simulation version of the design. I did not spend the time to merge the designs, so i had a executable version and a simulation version. And that was fine. I spent all day trying to rebuild it but still don’t see where else i can look.

Today, i cannot remember if you provided me the testbench file or you pointed me to it, or i am just confused.

Can you spend a few minutes guiding me to recover it?

Thank you

Jaime

Closed #7<https://github.com/fpgadeveloper/fpga-drive-aximm-pcie/issues/7>.

fpgadeveloper, 2020-06-09:
Hi Jamie,
I'm sorry but I don't have a testbench for simulating that design, and I've looked through my emails but there's nothing there regarding simulation.
Jeff

jdlr-at-nrec, 2020-06-09:
It was completely my confusion.

I profusely apologize for wasting your time. Much appreciate your effort, indeed.

Jaime