fpgadeveloper/fpga-drive-aximm-pcie #10: The board_part definition was not found for em.avnet.com:picozed_7015_fmc2:part0:1.1.

Bestduan, 2021-01-08:
[image]
[image]
excuse me, there is a question, how can i fix it?

fpgadeveloper, 2021-01-08:
Hi Bestduan,

You need to copy the board files into your Vivado installation. The instructions for this are here:

https://github.com/fpgadeveloper/fpga-drive-aximm-pcie#installation-of-picozed-board-definition-files

Also, I can see that you are using 2018.3 of the tools but we don't have a specific release for that version. Instead I recommend you use the 2018.2 release or the 2019.2 release: https://github.com/fpgadeveloper/fpga-drive-aximm-pcie/releases

We are currently working on the 2020.2 release and this should be available in the next few days.

Regards,
Jeff

closed 2021-01-08 by fpgadeveloper

Bestduan, 2021-01-08:
thank you, it works. About the version，i have rewrited the script. After that, in this project where call this board file, and where should i change in the board file, if i want use my own fpga board?