fpgadeveloper/fpga-drive-aximm-pcie #11: some problems about this script of 18.2 version

Bestduan, 2021-04-13:
Hello I found in reference for the design of [zynq-pci-express-root-complex-design-in-vivado](https://www.fpgadeveloper.com/2016/04/zynq-pci-express-root-complex-design-in-vivado.html/) and another project by using script to generate. After practice, I found that both projects have certain problems.
When using the the script of 18.2 version to generate the project, the first thing I encountered was that UART IP was not called. After I added UART IP, I found that the ELF file could not be downloaded, and the SDK reported an error showing address 0x8000_0000 download error. Considering that address edit may be a problem, I changed the address of BAR0 to 0x4000_0000, and vivado reported an critical warning again.
~~~
[BD 41-1288] 
Address segment &lt;/axi_pcie_0/S_AXI/BAR0&gt; in &lt;/processing_system7_0/Data&gt; at &lt;0x40000000 [ 256M ]&gt; is illegal. 
This address exceeds the base address limitations &lt;0x80000000 [ 1G ]&gt; of the interface(s) &lt;/processing_system7_0/M_AXI_GP1&gt; through which it is accessed by this address space.
~~~
In addition, I designed the program according to the Zynq-PCI-Express-Root-Complex-Design-in-Vivado tutorial, and the program downloaded normally, but when I executed the XAxiPcie_ReadReg function, it got stuck. How can I fix it ?

closed 2021-04-13 by Bestduan

neuhsm, 2022-05-07:
i meet the same problem with zc706 and vivado2018.3, how to make it work?

Bestduan, 2022-05-08:
https://github.com/Bestduan/pz_7z035_pcie
This is my project repo, you can refer to it.

neuhsm, 2022-05-08:
thank you for your replay，i find one thing really unexpected，maybe vivado2018.3 quite different from others,you may look at this,[https://support.xilinx.com/s/question/0D52E00006hpoxuSAA/zc706%E8%AF%84%E4%BC%B0%E6%9D%BF%E5%90%AF%E5%8A%A8%E5%86%85%E6%A0%B8%E5%BC%82%E5%B8%B8%E5%81%9C%E6%AD%A2?language=en_US](url)

Bestduan, 2022-05-08:
thank you for your replay, I will check it.

neuhsm, 2022-05-08:
sorry for my mistake,i'm new to zynq ,i find i use the mistake xdc file, i should use the zc706_hpc.xdc,because i use the fmc to connect with pcie,but i use the zc706.xdc which use the pcie finger to connect,so it of course goes wrong,i'm trying to make it work with the new xdc file,and really thanks for your reply in time.