fpgadeveloper/fpga-drive-aximm-pcie #12: VC707 and FPGA Drive SSD can't see console logs during the petalinux booting.

kmakhno, 2021-04-19:
Hello, we bought your FPGA Drive module and clone repo with project for Vivado 2018.2, but when I try to boot petalinux on vc707, I can't see any console logs. I've already see this issue https://github.com/fpgadeveloper/fpga-drive-aximm-pcie/issues/1#issue-239293179 but for me it's not working. But in qemu it works well. When I create simple project with microblaze and all prerequisites for petalinux I can boot linux and see console logs. May you have any suggestion to solve my problem?

fpgadeveloper, 2021-04-19:
Did you try to boot the standalone application? Does it work?
Here is the prebuilt standalone for VC707 HPC1:
https://opsero.com/downloads/boot/2020-2/vc707_hpc1_pcie-standalone-2020-2.tar.gz
Here is the prebuilt standalone for VC707 HPC2:
https://opsero.com/downloads/boot/2020-2/vc707_hpc2_pcie-standalone-2020-2.tar.gz

Did you try the latest version (Vivado 2020.2)?

I suggest you try to create a simple standalone project in Vitis using the Vivado project from this repo. Try a simple hello world application and see if it works.

kmakhno, 2021-04-19:
yes, I try to boot standalone app and it works. Our company is using vivado2018.2 for now( I thought that it should work.

kmakhno, 2021-04-19:
Also I did DRAM memory check and I have linked up pcie with standalone exam

fpgadeveloper, 2021-04-20:
Sometimes we have issues on older versions and they don't end up getting fixed.
Please try booting one of these PetaLinux builds for 2020.2
[Uploading v
[vc707-hpc2.zip](https://github.com/fpgadeveloper/fpga-drive-aximm-pcie/files/6344520/vc707-hpc2.zip)
c707-hpc1.zip…]()

. They are .mcs files for booting from flash.

fpgadeveloper, 2021-04-20:
[vc707-hpc1.zip](https://github.com/fpgadeveloper/fpga-drive-aximm-pcie/files/6344524/vc707-hpc1.zip)

kmakhno, 2021-04-20:
thanks for reply, I booted your configuration file but I've got this exception 
[image]

fpgadeveloper, 2021-04-20:
Strange. Are you using FPGA Drive FMC? What is the serial number?

kmakhno, 2021-04-20:
yes. I don't know what exactly serial number is)
[image]
[image]

fpgadeveloper, 2021-04-20:
That's OK I can see the serial number in your first photo.

Can you try booting PetaLinux from JTAG with these prebuilt files for 2020.2 and 2019.2:
https://opsero.com/downloads/boot/2020-2/vc707_hpc1_pcie-petalinux-2020-2.tar.gz
https://opsero.com/downloads/boot/2020-2/vc707_hpc2_pcie-petalinux-2020-2.tar.gz
https://opsero.com/downloads/boot/2019-2/vc707_hpc1_pcie-petalinux-2019-2.tar.gz
https://opsero.com/downloads/boot/2019-2/vc707_hpc2_pcie-petalinux-2019-2.tar.gz

To boot them, use XSCT and these commands:
```
  connect
  fpga -f system.bit
  target 3
  dow image.elf
  con
```
It will take a while to load the image.elf file because it is large. It will also take a couple minutes before you start seeing the boot log. Please try the 2020.2 version first and then the 2019.2 if that fails.

kmakhno, 2021-04-21:
hi, I tried it. At first I tried 2020 files and they gave me this logs
[image]
And I waited for some time but didn't get some logs
Then I tried 2019 files and I had the same situation as with 2018. After approximately 15 minutes I didn't get logs.