fpgadeveloper/fpga-drive-aximm-pcie #19: [ZCU106] Cannot detect the SSD

CaglayanDokme, 2021-12-07:
I followed the steps provided in the documentation and successfully generated the image files for the ZCU106 development board. The board boots up without any problem. But, somehow, I can't detect the SSDs. 

I faced the same problem with both of the following designs created for ZCU106

- ZCU106 HPC0
- ZCU106 HPC0 Dual

The output of the _lspci_ command for HPC0 Dual configuration is as follows:

> 0000:00:00.0 PCI bridge Xilinx Corporation Device 9134
> 0001:00:00.0 PCI bridge Xilinx Corporation Device 9134

The SSD part is one of the [tested ones](https://refdesign.fpgadrive.com/en/latest/supported_ssds.html#list-of-tested-ssds) by Opsero: Samsung 970 EVO 500GB

fpgadeveloper, 2021-12-07:
Hi,
Please try downloading the prebuilt boot files and testing them on your hardware. They have been tested on our hardware and with the same SSDs.
Dual design: https://opsero.com/downloads/boot/2020-2/zcu106_hpc0_dual-petalinux-2020-2.tar.gz
Single design: https://opsero.com/downloads/boot/2020-2/zcu106_hpc0-petalinux-2020-2.tar.gz
Please use the Opsero contact form for more support on this.
Jeff

closed 2021-12-07 by fpgadeveloper

CaglayanDokme, 2021-12-08:
Hi Jeff,
Thanks for the quick reply. I tried to regenerate the Vivado and PetaLinux projects using the scripts you've provided. The Vivado version I use is **2018.3** and the tickle script states that it should be **2018.2**. So, I upgraded the tickle file by just modifying the `version_required` variable. Would it cause a big difference between final products? I guess that the PetaLinux version also differs. Thus, I would be appreciated if you could tell me whether it would be a problem or not.

fpgadeveloper, 2021-12-08:
Contact me through the Opsero contact form.