fpgadeveloper/fpga-drive-aximm-pcie #20: Booting KCU105 HPC image without FPGA Drive

MasterJubei, 2022-07-20:
Is this supported? I want to see if the image will boot before getting a FPGA Drive. 
I generated the KCU105 image successfully with Vivado 2020.2

When booting I get stuck at this point:

```
Starting kernel ...

Ramdisk addr 0xaf7a4000,
FDT at 0xaf79d000
earlycon: ns16550a0 at MMIO 0x44a01000 (options '115200n8')
printk: bootconsole [ns16550a0] enabled
cma: Reserved 16 MiB at 0xae400000
Linux version 5.4.0-xilinx-v2020.2 (oe-user@oe-host) (gcc version 9.2.0 (GCC)) #1 Wed Jul 20 21:41:51 UTC 2022
setup_memory: max_mapnr: 0x7ffff
setup_memory: min_low_pfn: 0x80000
setup_memory: max_low_pfn: 0xb0000
setup_memory: max_pfn: 0xfffff
Zone ranges:
  DMA      [mem 0x0000000080000000-0x00000000afffffff]
  Normal   empty
  HighMem  [mem 0x00000000b0000000-0x00000000ffffefff]
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x0000000080000000-0x00000000ffffefff]
Initmem setup node 0 [mem 0x0000000080000000-0x00000000ffffefff]
setup_cpuinfo: initialising cpu 0
setup_cpuinfo: Using full CPU PVR support
wt_msr_noirq
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists, mobility grouping on.  Total pages: 522559
Kernel command line: console=tDentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
mem auto-init: stack:off, heap alloc:off, heap free:off
Memory: 2044036K/2097148K available (5556K kernel code, 182K rwdata, 1576K rodata, 170K init, 565K bss, 36728K reserved, 16384K cma-reserved, 1310716K highmem)
Kernel virtual memory layout:
  * 0xfffea000..0xfffff000  : fixmap
  * 0xff800000..0xffc00000  : highmem PTEs
  * 0xff7ff000xilinx_timer_init: Timer base: 0xd0020000, Clocksource base: 0xd0020010
clocksource: xilinx_clocksource: mask: 0xffffffff max_c/amba_pl/timer@41c00000: irq=2, cpu_id 0
xilinx_timer_shutdown
xilinx_timer_set_periodic
Calibrating delay loop... 49.35 BogoMIPS (lpj=246784)
pid_max: default: 4096 minimum: 301
Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Mountpoint-cache hash table entries: 2048 (order: 1, 8random: get_random_u32 called from bucket_table_alloc.isra.0+0x70/0x218 with crng_init=0
clocksource: jiffies: mask: 0xffffffffNET: Registered protocol family omic allocations
audit: initializing netlink subsys (disabled)
PCI: Probing PCI hardware
audit: type=2000 audit(0.290:1): state=initialized audit_enabled=0 res=1
vgaarb: loaded
clocksource: Switched to clocksource xilinx_clocksource
NET: Registered protocol family 2
tcp_listen_portaddr_hash hash table entries: 512 (order: 1, 12288 bytes, linear)
TCP established hash table entries: 8192 (order: 3, 32768 bytes, linear)
TCP Trying to unpack rootfs image as initramfs...
random: fast init done
Freeing initrd memory: 8556K
workingset: timestamp_bits=14 max_order=19 bucket_order=5
Key type cifs.idmap registered
romfs: ROMFS MTD (C) 2007 Red Hat, Inc.
bounce: pool size: 64 pages
io scheduler mq-deadline registered
io scheduler kyber registered
GPIO IRQ not connected
XGpio: gpio@40000000: registered, base is 511
```
Is this expected?
Thanks a lot

MasterJubei, 2022-07-21:
Actually I think it might still be working, it's just that the UART seems to stop working when Linux is booted up.

fpgadeveloper, 2022-07-21:
Hi,
Yes this design is still supported.
Unfortunately though, you wont be able to boot these designs without the FPGA Drive FMC installed. The reason is that the FPGA Drive FMC supplies a 100MHz clock that feeds the PCIe block, so the initialization of PCIe would fail without it.

closed 2022-07-21 by fpgadeveloper