fpgadeveloper/fpga-drive-aximm-pcie #28: Boot stall on VCK190

boolsmith, 2024-04-15:
I followed the guide to build this project for target `vck190_fmcp1`. Running: `make petalinux TARGET=vck190_fmcp1` works just fine, but when I program the board, the boot process stalls early on with no obvious errors:
``` bash
[0.015]****************************************
[0.070]Xilinx Versal Platform Loader and Manager
[0.126]Release 2022.1   Apr 11 2022  -  09:29:50
[0.183]Platform Version: v2.0 PMC: v2.0, PS: v2.0
[0.247]BOOTMODE: 0xE, MULTIBOOT: 0xF0000000
[0.303]****************************************
[0.521]Non Secure Boot
[3.399]PLM Initialization Time
[3.447]***********Boot PDI Load: Started***********
[3.507]Loading PDI from SD1_LS
[3.555]Monolithic/Master Device
[285.899]282.366 ms: PDI initialization time
[285.959]+++Loading Image#: 0x1, Name: lpd, Id: 0x04210002
[286.027]---Loading Partition#: 0x1, Id: 0xC
[337.869] 51.758 ms for Partition#: 0x1, Size: 2912 Bytes
[342.775]---Loading Partition#: 0x2, Id: 0x0
[347.209] 0.518 ms for Partition#: 0x2, Size: 48 Bytes
[351.404]---Loading Partition#: 0x3, Id: 0x0
[369.105] 13.783 ms for Partition#: 0x3, Size: 58912 Bytes
[371.409]---Loading Partition#: 0x4, Id: 0x0
[379.466] 4.139 ms for Partition#: 0x4, Size: 5888 Bytes
PSM Firmware version: 2022.1 [Build: Apr 11 2022 09:29:50 ]
[386.867]+++Loading Image#: 0x2, Name: fpd, Id: 0x0420C003
[391.919]---Loading Partition#: 0x5, Id: 0x8
[399.547] 3.711 ms for Partition#: 0x5, Size: 1568 Bytes
[404.000]+++Loading Image#: 0x3, Name: apu_subsystem, Id: 0x1C000000
[407.602]---Loading Partition#: 0x6, Id: 0x0
```
I have attempted to boot via SD card, and JTAG (using the same SD card image for rootfs), with the same results for either method.

NOTE: When I boot via jtag I use the command: `petalinux-boot --jteg --kernel`. I omitted the `--fpga` flag as described [here](https://refdesign.fpgadrive.com/en/latest/petalinux.html#id1) as I get the error message: "Invalid argument "--fpga" for system arch "versal""

I am running on Xilinx 2022-1 on a Ubuntu 20 machine.

fpgadeveloper, 2024-04-15:
Hi,

Could you try the pre-built boot files here:
[FPGA Drive FMC VCK190 FMCP1 pre-built boot files](https://download.opsero.com/boot/2022-1/fpgadrv_vck190_fmcp1_petalinux-2022-1.zip)

Have you followed the SD card preparation instructions here?
[Prepare the SD card](https://refdesign.fpgadrive.com/en/latest/petalinux.html#prepare-the-sd-card)

Does your board have production silicon or engineering sample silicon?

boolsmith, 2024-04-15:
Hi, 

I verified that the pre-built boot files can be used to successfully boot the board. 

I compared the SD card files I used with the supplied pre-built boot files to verify no differences. 

I have production silicone

I was able to get my project to successfully boot by starting from scratch and: making the [Vivado project](https://refdesign.fpgadrive.com/en/latest/build_instructions.html#build-vivado-project-in-linux), and the [Vitis project](https://refdesign.fpgadrive.com/en/latest/build_instructions.html#build-vitis-workspace-in-linux) before I made the [Petalinux project](https://refdesign.fpgadrive.com/en/latest/build_instructions.html#build-petalinux-project-in-linux). Based on the documentation, it sounds like making the vitis project is optional, and that when make is called in petalinux, it will call make for the Vivado project if it has not already been made. Is that correct? 

Of course, It could be that I was more careful the second time through making the project, so I will go back to verify that I built the original Petalinux project correctly.

fpgadeveloper, 2024-04-15:
Hi,

I'm glad that you have it working now.

If you only wish to use PetaLinux, then the Vitis project is not necessary - it is completely independent.

Yes, if you run the PetaLinux build before running the Vivado build, then the Vivado build will be run first. This is one of the benefits of using makefiles.

closed 2024-04-15 by fpgadeveloper